equinor/flotilla #1763: Missing translation

andchiind, 2024-10-14:
**Describe the bug**
This safezone error message is lacking a norwegian translation.

**To Reproduce**
Make the safezone mission fail.

**Expected behavior**
This should work in Norwegian and English.

**Screenshots**
[image]
[image]

Eddasol, 2024-10-24:
[image]
Same here

andchiind, 2024-10-25:
Same here [image]

andchiind, 2025-03-10:
No longer relevant

closed 2025-03-10 by andchiind