equinor/flotilla #1790: Multiple "drive to docking station" missions

Eddasol, 2024-11-12:
**Describe the bug**
When testing the "Send robots to dock" button several "drive to docking station" missions appeared. Sometimes the missions got stuck in the queue and didn't run even though the robots were dismissed from dock.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**

**Screenshots**
[image]

Eddasol, 2024-11-12:
Setting the status to available when the missions got stuck worked to get the missions to run again

andchiind, 2024-12-18:
We should test that this is still the case since the mechanics have recently been simplified.

tsundvoll, 2025-01-06:
Closing as this has not been a problem lately

closed 2025-01-06 by tsundvoll