equinor/flotilla #1820: Robot stuck after failed return to home

mrica-equinor, 2024-11-28:
**Describe the improvement you would like to see**
When sending a return to home mission, isar-anymal reported task as failed and as a consequence the return to home failed.

[image]

When this happens, if we try to reschedule a return to home by clicking the "send robots to dock", flotilla will check if the robot is already home first:

[image]

It sees that the last mission run was a return to home mission, regardless of weather or not it was successful and cancels the sheduling of the new return to home. 

[image]

So the robot gets "stuck".

andchiind, 2024-12-18:
This should no longer be the case, but we need to test it

andchiind, 2025-01-28:
Closing this. We can reopen it if we encounter a similar issue, but this should no longer happen

closed 2025-01-28 by andchiind