equinor/flotilla #1852: Remove the need or semaphores in mission scheduling

andchiind, 2024-12-06:
**Describe the improvement you would like to see**
Currently we use a semaphore for mission scheduling to manage the side effects of localisation and bulk scheduling of missions. We will now be removing the side effects of localisation checks when scheduling missions in Flotilla but the bulk scheduling of missions still poses a problem. To deal with this, we should ideally not create all missionRuns separately and then run the code to check whether to schedule the missions. Instead we should manage them together so that we keep track of which to schedule first and which to put on the queue. Concurrencly is handled by ISAR in the sense that if two requests to start a mission come at the same time, ISAR will only accept one. The thread managing the missionRuns where the missions is rejected by ISAR should handle this by instead putting the missionRuns on the queue.

**How will this change existing functionality?**
It will improve stability as semaphores don't work with multiple Flotilla instances running in Kubernetes.

**How will this improvement affect the current Threat Model?**
N/A

andchiind, 2025-01-10:
With the recent change where it will put a mission on the queue if ISAR is busy, I am not sure if the current use of the semaphore is needed. We could try test the code as is, but without the semaphore.

tsundvoll, 2025-01-13:
Needs to be tested if this is currently needed

closed 2025-01-13 by tsundvoll