equinor/flotilla #1878: Do not run "Send to dock" mission if the robot is docked

andchiind, 2024-12-12:
**Describe the improvement you would like to see**
If ISAR reports that the robot is already docked, we don't need to schedule a "send to dock" or "return home" mission. See https://github.com/equinor/isar/issues/680

**How will this change existing functionality?**
It is a minor change, but it will prevent us from needing to ignore illogical missions from Flotilla in ISAR.

**How will this improvement affect the current Threat Model?**
N/A

andchiind, 2025-07-03:
The solution I prefer is that we can trust that ISAR knows that it is home. Therefore ISAR should return a CONFLICT if it is in the home state when it receives return home, and then Flotilla can handle that as it likes.

andchiind, 2025-07-03:
Moving this to new to create discussion around my last comment.

mrica-equinor, 2025-07-07:
Related to: https://github.com/equinor/isar/issues/779

closed 2025-10-16 by andchiind