equinor/flotilla #1886: Return to home on low battery doesn't work

Eddasol, 2024-12-16:
**Describe the bug**
When the robot gets low battery it should cancel ongoing mission and return to home. Currently the robot will complete the ongoing mission (should be cancelled) and queue a return to home mission. The return to home mission doesn't start due to low battery (should start).

**To Reproduce**
Steps to reproduce the behavior:
* Let robot get low battery while running mission

**Expected behavior**
Should return home

**Screenshots**

Eddasol, 2024-12-16:
Related: https://github.com/equinor/flotilla/issues/1877

andchiind, 2024-12-18:
This needs to be tested again, might have been fixed in https://github.com/equinor/flotilla/pull/1889/commits/4cf6b3532e032975b40d2ee772570e6b15c1ae8d

andchiind, 2025-01-14:
It has indeed been fixed by the previously mentioned commit

closed 2025-01-14 by andchiind