equinor/flotilla #1887: Listen for a message of isar staring up

oysand, 2024-12-16:
**Describe the improvement you would like to see**
If ISAR sends a startup message, this could tell flotilla when ISAR have restarted. 

**How will this change existing functionality?**
Flotilla can now update mission status due to the isar instance have restarted

**How will this improvement affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

tsundvoll, 2025-01-13:
Maybe just use the fact that ISAR sends status "IDLE"; Flotilla can interpret this as "ongoing missions should be cancelled"

Eddasol, 2025-05-12:
Can end up in a state where:
* Flotilla have an ongoing mission that never will update and will be stuck in ongoing
* ISAR doesn't have any missions
* The robot is running a mission

Goal of the issue is to ensure that the mission in flotilla gets cancelled, and that the (potentially) ongoing mission on the robot gets the cancel message.

oysand, 2025-05-12:
When the status from isar is sent to flotilla and set to available, it happens at the same time as the mission finish. Based on which message arrives first there might be a time where there robot is available and the mission is ongoing, but this is fine as the mission update will get through shortly

Eddasol, 2025-07-10:
https://github.com/equinor/flotilla/pull/2268