equinor/flotilla #1888: Let ISAR generate IsarTaskId

tsundvoll, 2024-12-16:
Have the IsarTaskId for MissionTasks be generated by ISAR and reported back in the response when calling the start-mission endpoint in ISAR.

Similar to what is now done for IsarMissionId and IsarInspectionId.

This might require some refactoring, to have one object type be sendt to the ISAR insance (the mission request or specification), and one object instansiatied when the response comes back from ISAR.

Eddasol, 2025-06-18:
Suggest accept that isar uses the provided id and only generate new it none is received. This is the current implementation and the issue should therefor be considered closed as not planned

tsundvoll, 2025-06-25:
I agree to this approach, it simplifies it if Flotilla generates the IDs up front and ISAR just uses these. Then they can refer to the same things and not be confused.