equinor/flotilla #1955: Return to home mission is cancelled when running low on battery

andchiind, 2025-01-15:
**Describe the bug**
If we run low on battery we want to cancl the current mission and return home. However, this is being done even if the current mission is return home, which is not ideal. We only want to cancel non-return home or emergency missions, but doing this in a thread safe way may not be possible. 

**To Reproduce**
Have the robot become low on battery during return home mission.

**Expected behavior**
We should not cancel return home missions when running low on battery.

tsundvoll, 2025-01-27:
Not critical for functionality; and part of this may be moved to ISAR in the future, so setting it to TODO for now

andchiind, 2025-02-12:
This may already be fixed, so I'm moving it to the test column

andchiind, 2025-03-04:
It is still cancelling the return home and replacing it by an emergency return home mission, but this will soon be replaced by an ISAR return home solution. https://github.com/equinor/isar/issues/701

closed 2025-03-04 by andchiind