equinor/flotilla #1975: Return to Home task status doesn't update

mrica-equinor, 2025-01-24:
**Describe the bug**
The Return to Home Task Status is not updating:

[image]

Even though the robot successfully completed the Return to Home, the mission status updated to successfull but the task status doesn't. 

I assume this has to do with updating the task status on the OnIsarTaskReceived from the mqtt that is for some reason not writing to the database if the task type is return to home.

**To Reproduce**
Run a return to home mission.

**Expected behavior**
Task satus should change to ongoing and then successfull.

Eddasol, 2025-01-27:
Seems that the taskId received in "OnIsarTaskUpdate" is wrong for returnToHome missions, preventing flotilla from updating the task status

Eddasol, 2025-01-27:
Not prioritized as return to home will be refactored away by https://github.com/equinor/isar/issues/701

Eddasol, 2025-05-02:
Return to home missions no longer have a mission page. This is no longer relevant

closed 2025-05-02 by Eddasol