equinor/flotilla #1987: Missions get stuck in ongoing

Eddasol, 2025-01-28:
**Describe the bug**
Sometimes a mission gets status ongoing, but its not actually running and its not possible to stop/skip the mission as isar isn't running that mission. When one mission gets stuck in ongoing it will occupy the ongoing spot belonging to the robot card, which hides the actual ongoing mission.

**To Reproduce**
Steps to reproduce the behavior:
* Queue a large amount of (short) missions
* Wait for them to run
* Once in a while one of the missions get stuck in ongoing

**Expected behavior**
Only the ongoing mission should have status ongoing. There should max be one ongoing mission per robot

**Screenshots**
Not that time spent is 235minutes. Other missions have been both started and completed while this mission stayed ongoing. 
[image]

andchiind, 2025-02-11:
Has this been reproduced recently?

oysand, 2025-05-05:
Also happens when restarting ISAR during a mission #1943

Eddasol, 2025-05-05:
When this bug occurs one robot can end up with 2 ongoing missions. Seems that the mission that completed didn't get its state updated from ongoing to one of the completed states, even though it did complete. The mission isn't actually ongoing, but seems ongoing as it has an outdated state.

Eddasol, 2025-05-15:
[image]

closed 2025-09-08 by mrica-equinor