equinor/flotilla #2032: Remove Installation or Plant, or combine them

Christdej, 2025-02-18:
**Describe the improvement you would like to see**
Today we use Plant somewhere, and Installation somewhere. I dont believe we need both, so one should be removed. Also some models, like InspectionArea, contain both Plant and Installation, and as Plant also contain a Installation, it is possible to get into a situation where "Installation" is tracked multiple places at once.

Related issues:
https://github.com/equinor/flotilla/issues/876
https://github.com/equinor/flotilla/issues/1580

tsundvoll, 2025-05-12:
One Installation can have multiple Plants; we only need Plant (this implies installation)

andchiind, 2025-08-26:
> One Installation can have multiple Plants; we only need Plant (this implies installation)

A counter argument is that our access role system related to Installations, as does our filtering of robots and missions. It would be easiest to remove Plant. The only issue is that Echo predominantly uses Plants to filter missions, but this could be solved by storing missions in our own database instead of relying on Echo for this.

andchiind, 2025-09-22:
New update: Plant is useful for choosing which Echo map to use, so putting this issue on hold

andchiind, 2025-11-12:
The pain of making this change does not justify the benefit