equinor/flotilla #2040: Have flotilla read echo mission tasks as Application

mrica-equinor, 2025-02-21:
**Describe the improvement you would like to see**
Currently STID/Echo do not support calls to the API as application so we don't have a good way of having flotilla scheduling missions autonomously. 

Have a dialog with STID and ECHO on what teh best solution here is. 

Currently if the operator makes changes on a mission in the mission planner, it needs to manually fetch the mission from echo with the predefined mission button for the changes to take place in the auto scheduling of missions.

**How will this change existing functionality?**
Mission tasks are autonomously updated by flotilla when the mission is scheduled.

mrica-equinor, 2025-02-25:
Have asked ECHO Team to look into removing the call to STID from the endpoint. So we would get the STID coordinates for the target when creating/setting to ready a mission (since is this done by the user) and we would assume these coordinates don't change. 

Therefore when reading in the missions these positions would be stored already and no need to be calling the STID API.

tsundvoll, 2025-03-03:
Suggestion: We can notify the user in Flotilla if there is a difference between the scheduled missions and the updated missions in Echo.

oysand, 2025-05-06:
The auto schedule should read the echo mission from the source instead of the last successfull mission run before closing this