equinor/flotilla #2079: Add placeholder for no auto scheduling

Eddasol, 2025-03-13:
**Describe the improvement you would like to see**
When there is no auto scheduling the tab is empty. Should have a placeholder.
[image]

**How will this change existing functionality?**

**How will this improvement affect the current Threat Model?**

mariannewiggen, 2025-03-17:
Suggested solution

[image]

[image]

anetteu, 2025-05-14:
#2184 Adds placeholder and option to auto schedule missions also when no mission are auto scheduled. Closing this issue. New issue #2197 is created to follow up on the calendar suggestion

closed 2025-05-14 by anetteu