equinor/flotilla #2207: Return to home mission is scheduled when moving the Taurob arm

anetteu, 2025-05-20:
**Describe the bug**
A return to home mission is scheduled when moving the Taurob arm. Need to cancel the mission manually to make the robot available to do new arm movement. 

**To Reproduce**
Move arm of the Taurob robot

**Expected behavior**
No mission should be scheduled when moving the arm.

**Screenshots**
If applicable, add screenshots to help explain your problem.

tsundvoll, 2025-05-26:
This action could send ISAR to a "Maintenance" state. No return to home should be triggered from this state. The only way of getting it out of "Maintenance" state should be a user input.

anetteu, 2025-08-11:
@tsundvoll the new `intervention needed` state is similar to the `Maintenance` state you describe. Can it be used to solve this issue? @andchiind Maybe you are right that this can be solved in ISAR. Please create an issue.

andchiind, 2025-09-16:
The solution here is to remove moveArm missions