equinor/flotilla #2267: Show banner if return to home fails

Eddasol, 2025-07-04:
**Describe the improvement you would like to see**
Currently if a normal mission fails, a banner will be created. However no banner gets created if a return to home mission fails, as these missions aren't tracked in flotilla.

**How will this change existing functionality?**
Better information to user

**How will this improvement affect the current Threat Model?**

Eddasol, 2025-07-10:
Partially solved by #2284 that creates banner in scheduling of return home fails. https://github.com/equinor/isar/issues/842 will solve for return to home missions that fails after successful scheduling

akustn, 2025-07-16:
I will close this one as [equinor/isar#842](https://github.com/equinor/isar/issues/842) is done.

PR #2284 (2025-07-10): Add SignalRMessage for return-home-failure
## Ready for review checklist:
- [x] A self-review has been performed
- [x] All commits run individually
- [x] Temporary changes have been removed, like console.log, TODO, etc.
- [x] The PR has been tested locally
- [ ] A test have been written
  - [x] This change doesn't need a new test
- [x] Relevant issues are linked
- [x] Remaining work is documented in issues
  - [ ] There is no remaining work from this PR that require new issues
- [x] The changes does not introduce dead code as unused imports, functions etc.

Closes #2267
closed 2025-07-16 by akustn