equinor/flotilla #2271: Error handling for when a return to home mission fails and robot doesn't reach docking

mrica-equinor, 2025-07-07:
**Describe the improvement you would like to see**
better handling of when a robot fails a return to home mission should be implemented. We currently are implementing a banner stating that the mission has failed, but nothing else happens. 

Would be nice to have some kind of error handling with retrying to send the robot to dock for example.

**How will this change existing functionality?**
A clear and concise description on how this will affect current functionality.

**How will this improvement affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

andchiind, 2025-07-21:
Would be good to notify our team if it fails to return home. Potentially we can also retry before doing this. Whether the retry is in ISAR or in Flotilla is up to the person implementing, but personally I would prefer it in ISAR.

andchiind, 2025-07-30:
This sounds like a good idea. Currently we go from "returning_home" state to "robot_standing_still" if it fails.

[image]

I think instead this should transition to a new "error" or "intervention_needed" or something like that, state. Then we can automatically notify ourselves if it enters that state, and then we can add a new endpoint to the ISAR API to change it out of that state to "robot_standing_still" or "home".

mrica-equinor, 2025-08-01:
ISAR PR: https://github.com/equinor/isar/pull/870