equinor/flotilla #2329: Frontend: Only show/indicate valid choices for robot given installation and vice-versa

olaals, 2025-08-08:
**Describe the improvement you would like to see**
In "Predefined Mission" view in Flotilla frontend, it is perhaps better user experience if once the user has chosen a mission, only valid Robots are shown or somehow highlighted in the robot dropdown. 

It may be the other way around too if the user has chosen a robot too, so that only valid missions for that robot are shown. 

Essentially it should not be possible to select combination of mission and robots that give this error message after submitting

[image]

so when a user select for example this mission

[image]

only robots that are valid for the previous choice(s) should be shown in the next dropdown 

[image]

**How will this change existing functionality?**
The user is not able to select invalid combinations of missions and robots, and it is more clear during selection what combinations are valid before submitting.

**How will this improvement affect the current Threat Model?**
Does not affect threat model.

andchiind, 2025-08-26:
The reason this is difficult to do is that we are only showing the echo missions as we fetch them from the Echo API. We have therefore not processed them into Flotilla MissionDefinitions yet, and so we have not calculated which inspection area they belong to. We can try to process them each time we read them in from the database for the first time (we could check if we have already processed them before making a new entry), but this will make it much slower to process.

A long term solution could be to host the missions ourselves instead of using the Echo mission planner database. This way we could ensure that all the relevant information is stored in the database when the mission is first created.

oysand, 2026-02-09:
Closing, if the need arise in the future it can be revisited

closed 2026-02-09 by oysand