equinor/flotilla #2333: Extensive logging of token expired

Christdej, 2025-08-11:
**Describe the improvement you would like to see**
When token expires, we are logging alot of message like:

Lifetime validation failed. The token is expired.
Microsoft.IdentityModel.LoggingExtensions.IdentityLoggerAdapter

Try to get new token before logging.

tsundvoll, 2025-08-15:
There seems to always be five logs coming in a row. The original failure and 4 reconnect attempts with 0, 2, 10 and 30 seconds delay. This matches the default behavior of .withAutomaticReconnect()

[image]

tsundvoll, 2025-10-06:
Maybe related to Signal R

closed 2025-10-27 by Christdej